rocketscream/Low-Power #17: Conditions in which powerDown current is measured.

shirish47, 2015-10-07:
Hi, 
I am using this library with my own controller board and I am doing power down with ADC and BOD off but not getting anywhere close to 1.7uA I am still at minimun 500uA,... and Have nothing on IO port for this reading my voltage is 3.3 volts and also its is running at internal 1Mhz with 8Mhz/8 internal. 
So I am want to know exactly at what conditions this 1.7uA was measured... I am expecting to go below 100uA.

shirish47, 2015-10-07:
well I tested your code and it seems to work fine.. but when I use this library it does not work as expected.
with your example I see current to 23uA. and when I use in my code its 700uA.. :( seems to see what could be the cause.

shirish47, 2015-10-07:
PinChange Int consumes lots of current. I checked this out.

rocketscream, 2015-11-09:
Your board itself might have other quiescent current in any of the components like for example the regulator (the most common one). PinChange will consume more when compared to just edge interrupt without level detection.

closed 2015-11-09 by rocketscream