rocketscream/Low-Power #19: Time keeping in powerdown mode

shirish47, 2015-12-19:
I am using this library with time library (http://www.pjrc.com/teensy/td_libs_Time.html)
I make my code to sleep in powerdown for say 8 seconds and then check the time and its not exact.
timer0 is shutdown. Is it possible to keep timer0 on so the timekeeping is possible for TimeLib ?

shirish47, 2015-12-22:
ok I have figured out a way to use WDT timer's wakeup cycle as time keeping.. Once arduino is on timer works fine ..so it like 

.
.
.....
for(){
powerDown(sleep_8s);..
adjusttime(8);// adds 8 seconds 
if(some condition(period) or time) break;
}
...
.
.
This way we can maintain timer and also sleep for long..
once a day if I get the time from NTP Server it would give minimum error..

rocketscream, 2016-03-16:
Without Timer0 (which is off in power down mode), which your Time library uses to keep track of time will not work. Your approach to adjust might be okay with small drift over time.

closed 2016-03-16 by rocketscream