rocketscream/Low-Power #21: Feature Request: Custom WDT handler

KaloNK, 2016-03-13:
Hello,
 I need to do some more stuff inside the WDT ISR handler, but want to avoid modifying the library directly.
 Can you please add and ifdef for CustomWDThandler or some other way to allow external function to be called. Something like below?
```
#ifdef CustomWDThandler
 CustomWDThandler;
#else
 wdt_disable();
#endif
```

rocketscream, 2016-04-01:
I would have to say no to this as not that many people would want to murk around with WDT unless they know what they are doing. And the only thing I would suggest in an ISR routine is setting or clearing a flag/semaphore, not more than that.

closed 2016-04-01 by rocketscream

jk987, 2017-10-23:
Well, so now we are two of us who would like to have this possibility.

And as @rocketscream was saying I needed it for setting or clearing one flag.

I "solved" it by copying the powerDown() code (and the dependecies) into my own library.

Btw, as I'm thinking about it I needed to have a flag inside WDT ISR only to recognize whether Arduino was woken by WDT or eg. serial line. Maybe I wouldn't need having my own WDT ISR if powerDown() had been returning true or false, whether it slept all the requested time (ie. WDT ISR was executed) or no.

jk987, 2017-10-23:
Draft of my idea:

LowPower:
```
volatile bool _woken_by_wdt;
bool LowPowerClass::powerDown(...) {
  _woken_by_wdt = false;
  ...
    lowPowerBodOnOrOff(...);
  ...
  return _woken_by_wdt;
}

ISR (WDT_vect) {
  wdt_disable();
  _woken_by_wdt = true;
}
```
User code:
```
if (LowPower.powerDown(...))
  // I was sleeping all the time.
else
  // I was woken up by something else, eg. some digital input.
```

ortegafernando, 2017-10-23:
I am also interested in a "special WDT_vect". I have finally delete the library's definition of ISR(WDT_vect) and defined my own in my sketch
The primary objetive is to have an option in my sketch to sleep the ATMEGA and also to have an extend watchdog reset protection of more than 8 seconds.

@rocketscream may be you know how to define "wake" your ISR(DWT_vect) in your library, so anyone than wants to override your function could do it in his sketch

This way, there vill be no problem with your library and also we could change the behaviour of the ISR(WDT_vect).

is it possible? 

regards,