rocketscream/Low-Power #22: Arduino 1.6 support (?)

rvalle, 2016-03-15:
Hi!

We have based our HW design on your  Arduino Ultra for a couple of years now.

We are producing a new board upgrade, So I am currently testing low power.

I noticed that the maximum low power mode (1.7 uA) can be achieved with Arduino 1.0.

However, with Arduino 1.6, with the same sketch, library and HW it measures 20.5 uA instead.

For testing I use the external interrupt example provided with the library. Slightly modified by enabling the internal pull-up resistor on the wakeup pin, or the Interrupt routing keeps getting called.

On my board, I only have the ATMEGA 328P, Resonator, Regulator, and the few passives required. 

I would say some flag is not getting passed or ignored in Arduino 1.6? I thought I would ask before looking into it...

Have you noticed this? or is it just me?

Rafael

rocketscream, 2016-03-16:
Hi Rafael,

I notice this while testing a batch of Mini Ultra 8 MHz last week. This is due to compiler optimization that basically leaves the BOD not properly configured (enabled!). Fixed in latest V1.50 release. Tested on both 1.0.x and 1.6.x branch of the Arduino IDE. Thank you for reporting.

closed 2016-03-16 by rocketscream

rvalle, 2016-03-16:
Thanks! great!