rocketscream/Low-Power #25: Feature Request: ESP8266 Support

TheY3TI, 2016-05-18:
Hi!

First of all, this library is great! It greatly reduces power consumption without any real effort.
That's why I'm interested in something similar for the now very popular ESP8266 which is seriously power hungry. I think you'd make a lot of people very happy.

Thanks for considering and keep up the good work!
Brent

rocketscream, 2016-05-19:
I believe it has been implement on the ESP8266 core already? Take a look [here](https://github.com/esp8266/Arduino/blob/master/doc/libraries.md#esp-specific-apis). The ESP8266 has more consideration before going into sleep as it has a full stack implementation of WiFi running on it.

closed 2016-05-19 by rocketscream

TheY3TI, 2016-05-19:
Indeed it has been, thank you very much for the quick answer!