rocketscream/Low-Power #30: ATMega2560 - power_usart3_disable' was not declared in this scope

J-M-L, 2016-11-06:
Hello 

```
#include "LowPower.h"
void setup()
{
  Serial.begin(115200);
}

void loop()
{
  Serial.println("Bye for 2s");Serial.flush();
  LowPower.powerDown(SLEEP_2S, ADC_OFF, BOD_OFF);
  Serial.println("Hi");Serial.flush();
}
```

compiles fine with an Arduino Uno in Version 1.6.12 of the IDE.

But moving this simple code to an Arduino Mega 2560 and trying to compile gets me an error:

```
.../Arduino/libraries/Low-Power/LowPower.cpp: In member function 'void LowPowerClass::idle(period_t, adc_t, timer5_t, timer4_t, timer3_t, timer2_t, timer1_t, timer0_t, spi_t, usart3_t, usart2_t, usart1_t, usart0_t, twi_t)':
.../Arduino/libraries/Low-Power/LowPower.cpp:440:49: error: 'power_usart3_disable' was not declared in this scope
  if (usart3 == USART3_OFF) power_usart3_disable();
                                                 ^
```

seems this was already reported [https://github.com/arduino/Arduino/issues/5286]()

bsoufflet, 2017-01-31:
+1 also on LowPower version 1.60 and arduino IDE 1.8.1 + Arduino Mega : 

> LowPower.cpp: In member function 'void LowPowerClass::idle(period_t, adc_t, timer5_t, timer4_t, timer3_t, timer2_t, timer1_t, timer0_t, spi_t, usart3_t, usart2_t, usart1_t, usart0_t, twi_t)':
> LowPower.cpp:440:49: error: 'power_usart3_disable' was not declared in this scope
>   if (usart3 == USART3_OFF) power_usart3_disable();
>                                                  ^
> LowPower.cpp:475:48: error: 'power_usart3_enable' was not declared in this scope
>   if (usart3 == USART3_OFF) power_usart3_enable();
>                                                 ^

Cysign, 2017-02-17:
Same here. Did anyone fix this?

Cysign, 2017-02-17:
By removing line 440 and line 475 I was able to compile properly.

line 440:
	if (usart3 == USART3_OFF)	power_usart3_disable();

line 475:
	if (usart3 == USART3_OFF)	power_usart3_enable();

AnoxySoftware, 2017-10-16:
A better solution is to use a preprocessor macro in case this get's added back someday in the power.h of avr:

```
	#if defined(__AVR_HAVE_PRR_PRUSART3)
	if (usart3 == USART3_OFF)	power_usart3_disable();
	#endif
```
```
	#if defined(__AVR_HAVE_PRR_PRUSART3)
	if (usart3 == USART3_OFF)	power_usart3_enable();
	#endif
```

rocketscream, 2017-10-16:
@AnoxySoftware thank you for the suggestion. I'm going to do a clean up on the libraries.

tvixen, 2018-06-13:
Arduino Mega 2560 issue still not fixed.  

In Arduino Compiler 1.8.5 we get this error:
LowPower.cpp:597: error: 'power_usart3_enable' was not declared in this scope
  if (usart3 == USART3_OFF) power_usart3_enable();
exit status 1
'power_usart3_disable' was not declared in this scope

For your information, its working with Nano, and Uno.

maxholgasson, 2018-07-09:
Same here. Still using 1.8.4

CaptChilko, 2019-05-05:
Unfortunately still not fixed, probably not good to have marked as closed seeing as the readme states this board is supported.