rocketscream/Low-Power #31: usb is closed ............?

iqbalpalemad, 2016-12-15:
Hello 

i have uploaded the idleWakePeriodic example on my  BlueDuino . which has a ATMega32U4 processor.
i have commented the  line for ATMega32U4 in the example and uploaded the sketch. the sketch was uploaded successfully. But after that my pc is not recognizing the board. Which shows usb device not recognized . i think the usb close command is closed the usb function of board . so now how can i get back to my previous stage. now i cant upload any sketch...........?

neolao, 2016-12-15:
When you press the reset button, you have a short time during which you can upload sketch

iqbalpalemad, 2016-12-15:
but the usb device is not detecting to upload sketch

iqbalpalemad, 2016-12-15:
have any other method.
i am un able to upload sketch. the port is not shown in arduino IDE

neolao, 2016-12-15:
you can set the port manually in file preferences.txt

iqbalpalemad, 2016-12-15:
can you please explain how to do it...?

the device is not shown in device manager

neolao, 2016-12-15:
What is your system?

iqbalpalemad, 2016-12-15:
Lenovo G510 with windows 10 os
i am using arduino IDE 1.6.2
the board used is BlueDuino

iqbalpalemad, 2016-12-15:
Lenovo G510 with windows 10 os
i am using arduino IDE 1.6.12
the board used is BlueDuino

neolao, 2016-12-15:
Ah ok, I use MacOS and Linux.

I think on Windows, you just have to find the right path to you physical USB port (the path you could select on IDE), maybe `COME6`.

[image]

Close the IDE and you set in the file found here:

[image]

iqbalpalemad, 2016-12-15:
[image]

on my device manager the board is shown like this 

[image]

and no port is shown in arduino ide

neolao, 2016-12-15:
Do you remember what was the port?

My solution is to not use the IDE to select the port.

iqbalpalemad, 2016-12-15:
the port was com6

rocketscream, 2017-03-14:
Always add a significant amount of delay at the start of the sketch to allow you the chance to upload code subsequently. ATMega32U4 still lacks a proper USB.detach() function call that supposed to be call before sleep mode is entered.

closed 2017-03-14 by rocketscream