rocketscream/Low-Power #34: Low Power external interrupt error between two processors

stakle, 2017-03-23:
Hi,
I am using the Low Power example library as directed by you in the externalinterrupt example. 

I used it on the Arduino UNO (328P) and it worked perfectly with my pushbutton w/ resistor and capacitor pulldown pair for the RISING interrupt configuration. (i.e. when OFF, the interrupt was pulled down to GND and then with the button pushed it rose to 5V.) This worked perfectly on the UNO when using digital pin 2 (i.e. attachinterrupt(0))

when I try to port this logic to the Arduino Micro (32U4), the same configuration but this time at the external interrupt at digital pin 7 on the Micro, it fails to wake the arduino micro up. I have checked and re-checked to be sure that I'm not using anything else on that pin and also don't have anything changed between the set-ups other than the pin number.

I could really use the help, am I missing some critical difference between the 32u4 and 328P that is not the different pins that allow for external interrupts.

P.S.
-I have also tried to use the different trigger modes and none of them seem to work.

-I have put an indicator LED to make sure that the voltage actually does go up when I press the button.

jozef, 2017-04-06:
experimented with Arduino Micro (32U4) and interrupt signal on pin 7.

`LowPower.powerDown(SLEEP_8S, ADC_OFF, BOD_OFF);` will sleep those full 8s and the external signal on pin 7 won't do anything. Replacing the LowPower.powerDown() line with `LowPower.idle(SLEEP_8S, ADC_OFF, TIMER4_OFF, TIMER3_OFF, TIMER1_OFF, TIMER0_OFF, SPI_OFF, USART1_OFF, TWI_OFF, USB_OFF);` will work es expected → sleep for 8s unless there is an external signal that will interrupt it.

jozef, 2017-04-09:
switched to using pin 0 and 1, and interrupts on these both work just fine and do wake-up micro for LowPower.powerDown() sleep.

closed 2018-09-14 by rocketscream