rocketscream/Low-Power #43: Sleep for 4 hours

josephernest, 2017-09-22:
Congrats for this great library!
I'd like to sleep the 328p for 4 hours. How to do that in order to use the least power possible ( < 0.01 mA) ?

rocketscream, 2017-09-23:
You can do something like this:

```
// 4 hours = 60x60x4 = 14400 s
// 14400 s / 8 s = 1800
unsigned int sleepCounter;
for (sleepCounter = 1800; sleepCounter > 0; sleepCounter--)
{
  LowPower.powerDown(SLEEP_8S, ADC_OFF, BOD_OFF);  
}
```
Please use the [forum](http://www.rocketscream.com/blog/forums/forum/libraries/low-power/) next time for questions on the library. The repository is for issues and bug reporting.

closed 2017-09-23 by rocketscream

josephernest, 2017-09-24:
Thanks @rocketscream, I already thought about this, but this will require 1800 sleep / wakeup / sleep / wakup ... operations, and I think it's not optimal. The goal is to reach < 10 µA [as mentioned here](https://electronics.stackexchange.com/a/128685/43434) during 6 hours.

rocketscream, 2017-09-24:
No, the time is limited by what the watchdog timer can provide which is 8s maximum. And the wake up time is only a tiny fraction compared to the 8s, in few us. So, will be still very low.

jassing, 2017-09-24:
You could use some sort of external trigger to wake the board up.    While
I am no 555 expert, I'm sure there's some way to use that to generate your
sleep duration; but I'm not sure it would use less power than waking the
board up every 8 seconds.   If I may ask, why the requirement?   If you're
trying to run off an AA battery, there are some very small lipo batteries
that carry a lot more power...

‌

dolanmiu, 2017-10-03:
Does this method mess up the Arduino clock at all?

I want it to sleep for 4 hours+, but I also have alarms which must activate at certain times

jassing, 2017-10-03:
Sounds like conflicting requirements.

dolanmiu, 2017-10-03:
@jassing 

This guy manages to sort of do it:

https://github.com/rocketscream/Low-Power/issues/19

vazquezjm, 2019-09-02:
@dolanmiu were you able to sort it out?

dolanmiu, 2019-09-09:
Nope never managed to do it

Ended up sleeping for 8 seconds instead

nimasaj, 2019-09-09:
You have to define a loop for it to iterate in the loop as long as you want. E.g 4 hours= 14400 seconds and 14400/8=1800. So, your loop should iterate for 1800 times, to be in (almost) sleep mode for 4 hours.

vazquezjm, 2019-09-09:
> You have to define a loop for it to iterate in the loop as long as you want. 
Right, as explained here: 
https://github.com/rocketscream/Low-Power/issues/43#issuecomment-331614084