rocketscream/Low-Power #45: SLEEP_MODE_EXT_STANDBY not declared on ATMega168

Erik84750, 2017-10-11:
When compiling with LowPower.h using Arduino Pro Mini 168 there is still an error present. Same error was reported in July 30 2015 (issue #14).

[code]

In file included from C:\Users\Mysafety\Google Drive\documents\Arduino\libraries\Low-Power\LowPower.cpp:32:0:

C:\Users\Mysafety\Google Drive\documents\Arduino\libraries\Low-Power\LowPower.cpp: In member function 'void LowPowerClass::powerExtStandby(period_t, adc_t, bod_t, timer2_t)':

C:\Users\Mysafety\Google Drive\documents\Arduino\libraries\Low-Power\LowPower.cpp:980:18: error: 'SLEEP_MODE_EXT_STANDBY' was not declared in this scope

    lowPowerBodOn(SLEEP_MODE_EXT_STANDBY);

                  ^

C:\Users\Mysafety\Google Drive\documents\Arduino\libraries\Low-Power\LowPower.cpp:980:4: note: in expansion of macro 'lowPowerBodOn'

    lowPowerBodOn(SLEEP_MODE_EXT_STANDBY);

    ^

C:\Users\Mysafety\Google Drive\documents\Arduino\libraries\Low-Power\LowPower.cpp:985:17: error: 'SLEEP_MODE_EXT_STANDBY' was not declared in this scope

   lowPowerBodOn(SLEEP_MODE_EXT_STANDBY);

                 ^

C:\Users\Mysafety\Google Drive\documents\Arduino\libraries\Low-Power\LowPower.cpp:985:3: note: in expansion of macro 'lowPowerBodOn'

   lowPowerBodOn(SLEEP_MODE_EXT_STANDBY);

   ^

Bibliotheek Wire op versie 1.0 in map: C:\Users\Mysafety\Google Drive\documents\Arduino\libraries\Wire  wordt gebruikt
Bibliotheek RTClibExtended op versie 1.0.0 in map: C:\Users\Mysafety\Google Drive\documents\Arduino\libraries\RTClibExtended  wordt gebruikt
Bibliotheek Low-Power op versie 1.6 in map: C:\Users\Mysafety\Google Drive\documents\Arduino\libraries\Low-Power  wordt gebruikt
exit status 1
Fout bij het compileren van board Arduino Pro or Pro Mini

[/code]

Erik84750, 2017-10-11:
I solved it by adding to LowPower.h the following:

#if defined (__AVR_ATmega168__)
		#define SLEEP_MODE_EXT_STANDBY SLEEP_MODE_STANDBY
	  #endif

cjander, 2017-10-12:
since you have a solution already, why not just create a pullrequest?

Erik84750, 2017-10-12:
I am sorry!! I am new to this, I have to learn how to use Github.
One additional thing I have to mention is that this relates to Atmega 168P, not the plain version.

rocketscream, 2017-10-12:
Actually, the non-P version of the ATmega168 does not have extended power save mode unlike the ATmega168P. If you do that, it will result in error as those bits are not present in the register. I think I would have to allow it to compile for P version only when it comes to that sleep mode and provide warning for non-P.

Erik84750, 2019-01-19:
Hi rocketscream, is it possible you had added this to your low power library? When I open LowPower.h I cannot find my original addition, but it seems replaced by equivalent code?