rocketscream/Low-Power #50: support for atmega328PB

VojislavM, 2017-12-21:
Is it possible to use this code with new atmega328PB?
http://www.microchip.com/wwwproducts/en/ATmega328PB

glaucomvp, 2018-04-10:
I'm also trying to but without success.

canique, 2018-04-28:
No it's not possible as of now.

I have personally rewritten parts of the code but there are issues:
1) the lowpower lib references avr/power.h - power_spi_enable()/power_twi_enable() are not defined there for the Atmega328PB.

2) the powerdown mode severely influences the ability to reset the device. I have not found the cause yet, but when I use powerdown mode in a sketch and try to flash via FTDI - avrdude gets stuck.
Even if I'm able to reset, flashing a new sketch won't work. So there is something peculiar about powerdown mode...

rocketscream, 2018-05-01:
Support on the B version is yet to be done. Do take a look at the [differences here](http://www.microchip.com//wwwAppNotes/AppNotes.aspx?appnote=en591418).

joysfera, 2019-06-23:
Here's the link to Canique's fork of this library:
https://github.com/canique/Low-Power-Canique

Power Down works well for me on 328PB, but I'm having issues with Power Save mode - it's using too much current (500 µA with Timer2 running synchronously with 8 MHz internal RC on 3 V). I believe the PTC needs to be disabled before entering Power Save mode, but it's not obvious how to do it. Yes, you can disable power to PTC module but is that the right way of disabling it? Compare with ADC - it needs to be disabled using ADCSRA...