rocketscream/Low-Power #52: Pro Mini still consuming 5-10ma?

solars, 2018-02-03:
Hi, I need help - my pro mini setup uses 5-10ma in sleep mode with the Lowpower lib, and I don’t know why. Maybe someone spots an error. I'm using a sample from TTN (Lora Network).

This is my setup:

Arduino Pro Mini (without LED) 8mhz/3.3v
Directly connected RFM95

Controlled through transistor (Pin 7 in the sketch)
Voltage Regulator + Level Converter + HC-SR04

This is the sketch:

https://gist.github.com/solars/29f523c5ad2a931b54d7003787f8e6f3

Am I doing it wrong?

ortegafernando, 2018-02-03:
Hi, are you powering it with 3.3v directly to the 3.3V pin ???? other way, if you are using the voltage regulator of the pro mini, probably it is what is consuming 4-5 mA.

solars, 2018-02-03:
@ortegafernando thanks for the quick reply! You could be right, I'm not sure where I connected it - will check

solars, 2018-02-07:
I found it, the reason was that since I'm switching GND with the transistor, VCC apparently connects to GND through the Trigger Pin I use for the HC-SR04 - which is set to LOW.

If I switch it go HIGH before sleeping, the consumption goes down to 78ua for the whole system.

closed 2018-02-07 by solars