rocketscream/Low-Power #62: Cannot use radio transmitter after switching to LowPower library

ribtoks, 2018-08-03:
My Arduino Mini board has DHT22 temperature sensor and 433MHz radio transmitter modules. I'm using `DHT` and `RCswitch` libraries to make use of them. Code looks like this:

    void setup() {
      tempSwitch.enableTransmit(RADIO_PIN);
      dht.begin();
    }

    void loop() {
      float temp = dht.readTemperature();
  
      unsigned long message = createTemperatureMessage(SENSOR_ID, temp);
      tempSwitch.send(message, 32);

    #ifdef MY_DEBUG
      unsigned int sleepCount = 1; 
    #else
      // 3600s / 2 / 8s == 225
      unsigned int sleepCount = 225;
    #endif
      for (; sleepCount > 0; sleepCount--) {
        LowPower.powerDown(SLEEP_8S, ADC_OFF, BOD_OFF);
      }
    }

So I'm reading temperature and sending it using `RCswitch` library. Initially I just had `delay(1000);` instead of fancy `LowPower` sleep loop but I wanted to make my "device" more battery-friendly so I switched to low-power sleep.

The problem is that now Arduino is only sending data first time and after sleeping 30 minutes (in "release" mode) it does not send anything anymore. If I reduce 30 minutes to 8 seconds it is capable of sending data as well as if I replace sleep with `delay(30*60*1000);`.

From beyond it looks like something is not caching up after the sleep. Do I need to "wake" Arduino in any special way? How to fix that?

Thanks

lukas12p, 2018-08-03:
Try do **float temp = dht.readTemperature();** twice with 1000msec delay to wake up dht.

ribtoks, 2018-08-03:
@lukas12p I will try, but so far I assume reading temperature works, but sending data with radio does not work. I log all radio signals on the receiver side so if the temperature reading would fail - it would just log some garbage.

Maybe I just need to insert `delay(1000);` after power sleep to wake up everything?

rocketscream, 2018-08-03:
Yes, you should dd the delay(1000) after sleep and probably before sleep too. I believe the DHT sensor needs some time to stabilize after waking up. You can also replace the transmission data with a known constant to prove this.

ribtoks, 2018-08-03:
@rocketscream will try, thanks!

ribtoks, 2018-08-03:
@rocketscream do I need to "reinitialize" radio module and DHT22 after power-down sleep? Is it equal for them as if you would just turn them off? I mean do I need to rerun `setup()` after low-power sleep?

rocketscream, 2018-08-03:
I have never use that radio module nor the DHT22 before but by right it shouldn't and will not mess anything related to it by going to sleep. Have you try removing the sensor portion and simply sending known constant data with the radio?