rocketscream/Low-Power #63: ISRs - Does the code run when waking up?

alc6379, 2018-08-15:
Device: Adafruit Feather 32u4

I want to use this library to not only wake up the device, but I wanted the ISR to actually do something when triggered. The example provided has the ISR as a no-op. 

I just want to set a volatile boolean as true when the ISR is triggered. Is this possible? Or does the interrupt just wake the MCU up and return control to the loop() function?

rocketscream, 2018-09-14:
You should always just set a flag in the ISR routine and process them in the main loop. The flag must be set as volatile else compiler might optimize them. Again this is application specific and not supposed to be handled by the library.

closed 2018-09-14 by rocketscream

alc6379, 2018-09-14:
Right-- that's all I'm doing in this particular case, like I said in my post. It just didn't seem like the flag was being set, even though the ISR had that code in it.