rocketscream/Low-Power #69: LowPower.powerDown not realiable over extended periods of time (ATmega328)

Rumidom, 2018-09-18:
Im using this snippet to make an Arduíno pro mini sleep for one hour  by setting sleepTime = 3600, it works fine for about 28 days and then it starts to to bypass the function altogether, after a couple of hours it disables the ADC and keeps going

` 
void loop() {
for(int j = 0; j < (sleepTime/8); j++)  {
      LowPower.powerDown(SLEEP_8S, ADC_OFF, BOD_OFF);
 }
resetFunc(); 
}

void(*resetFunc) (void) = 0;
`

rocketscream, 2018-10-04:
Is there all in your code? 
Usually I would suspect usage of wrong variable type or somewhere else you are accidentally using the variable for counting.

Rumidom, 2018-10-05:
I'm running a test where i keep printing the Arduino's  available memory, if there is a memory leak over time i should see it, im 8 days in to the test, and so far memory usage remained constant. i think it has something to do with the function 'resetFunc'  i'm using,  or maybe the Arduino i was using got defective somehow,  i'll try connecting the reset pin of the Arduino to an IO pin, and reset it that way,