rocketscream/Low-Power #70: Watchdog timer clash with other libs using ISR(WDT_vect)

systemofapwne, 2018-10-01:
When using this library in conjunction with any other library, which somehow seems to interact with 
watchdog timers, the compile fails. 

`libraries\Low-Power-master\LowPower.cpp.o (symbol from plugin): In function `LowPowerClass::idle(period_t, adc_t, timer2_t, timer1_t, timer0_t, spi_t, usart0_t, twi_t)':

(.text+0x0): multiple definition of `__vector_6'

libraries\Crypto\RNG.cpp.o (symbol from plugin):(.text+0x0): first defined here

collect2.exe: error: ld returned 1 exit status`

I narrowed it down to be a problem with definining ISR in multiple contexts.

This lib:
https://github.com/rocketscream/Low-Power/blob/master/LowPower.cpp#L1131
libcrypto:
https://github.com/rweather/arduinolibs/blob/master/libraries/Crypto/RNG.cpp#L228

Is the use of ISR necessary to work, if I do not want to disable the watchdog-timer?
If not, I suggest to add an optional preprocessor directive to exclude this code to avoid multiple definitions.

rocketscream, 2018-10-01:
Yes, the watchdog is used as a timer to wake the MCU up. I'm not sure whether you can avoid this because both are accessing the same resource here.

systemofapwne, 2018-10-01:
For now, I disabled the WDT in the crypto-lib, which makes the RNG for my crypto-project basically useless (the WTD-jitter has been used to produce entropy for seeding the RNG).
Can I safely disable the WDT in the low-power lib, if I only use SLEEP_FOREVER or will something else break?

Cheers

rocketscream, 2018-10-01:
If you use SLEEP_FOREVER, the WDT code section won't be called. You could do that. Maybe an `#ifdef `would help.

closed 2018-10-04 by rocketscream