rocketscream/Low-Power #75: first 9 cycles are not working

dmnc-net, 2018-10-25:
Hi, I need some help with finding the issue. I want to put my board to sleep for a bigger amount of time so I've created a function which repeats 8sec sleep command. In the example code below everything is working but in my complex project the problem occurs. First 9 cycles doesn't sleep and after that 10th call is working. Any clue what can affect this? Thanks.

```
#include <LowPower.h>

void lowPower(uint16_t len){
    for (uint8_t i = 1; i<=round(len/8+0.5); i++) {
      Serial.print(i);
      Serial.flush();
      LowPower.idle(SLEEP_8S, ADC_OFF, TIMER2_OFF, TIMER1_OFF, TIMER0_OFF, SPI_OFF, USART0_OFF, TWI_OFF);
      Serial.print("/");
      Serial.flush();
    }
}

void setup() {
  Serial.begin(38400);
  lowPower(30);
}

void loop() {
}

```

dmnc-net, 2018-10-25:
okay
1) number 9 is not solid
2) this issue is caused by this function which does power on/off for the GSM/GPS board with SIM808:
```
#define MODEM_PWR_PIN               9
#define MODEM_PWR_IMPULSE           1500

void ATHelper::powerImpulse() {
  pinMode(MODEM_PWR_PIN, OUTPUT);
  digitalWrite(MODEM_PWR_PIN, HIGH);
  delay (MODEM_PWR_IMPULSE);
  digitalWrite(MODEM_PWR_PIN, LOW);
}

```

closed 2018-10-27 by dmnc-net

fsantanna, 2018-10-27:
, to

Closed #75 <https://github.com/rocketscream/Low-Power/issues/75>.

<https://github.com/rocketscream/Low-Power/issues/75#event-1930440809>, or mute
the thread