rocketscream/Low-Power #78: Setting SAMD21 to Sleep (idle) is not doing anything

Jackjan4, 2019-01-23:
Hi,

when I use the library for an SAMD21 (Adafruit Feather M0) to put the SAMD21 in one of its idle modes (Idle 0 to idle 2) its not going into idle mode. It just executes the code after the idle code directly instead of listening on external interrupt. However, if I use standby (LowPower.standby()) it is working fine.

Jackjan4, 2019-01-27:
Further investigation showed that this was obviously due to the fact that interrupts are still called from other places within the system. So this is not a problem of LowPower.

closed 2019-01-27 by Jackjan4

jsiddall, 2021-03-22:
In case anyone comes across this searching, as I did, it is possible to turn off systick which prevents idle from returning every millisecond:

```
SysTick->CTRL &= ~SysTick_CTRL_TICKINT_Msk;
LowPower.idle(IDLE_2);
SysTick->CTRL |= SysTick_CTRL_TICKINT_Msk;
```