rocketscream/Low-Power #80: Compile problem: multiple definition of `__vector_6'

andreas-bulling, 2019-02-10:
```
libraries/Low-Power-master/LowPower.cpp.o (symbol from plugin): In function `LowPowerClass::idle(period_t, adc_t, timer2_t, timer1_t, timer0_t, spi_t, usart0_t, twi_t)':
(.text+0x0): multiple definition of `__vector_6'
sketch/sensor_mysensor.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Arduino Pro or Pro Mini.
```

rocketscream, 2019-02-17:
Are you using any other library that possibly accessing the same interrupt vector?
What version of the Arduino AVR board support are you using?

bkrajendra, 2020-02-14:
Hi
I am using this with MySensor Library which might be using Interrupts.
any possible solution to resolve this issue.?

HiFiJ, 2020-03-18:
Hi everyone,

I get the exact compilation error when trying to use it with MySensors 2.3.2.
Is there any workaround possible?
Thanks very much for your efforts,
Joost

cYnd, 2020-03-19:
Hi all,

I want to use this library with the radiohead encrypted rf95 server and I am running in the same problem. Is there any possible workaround?

eldiamond2, 2020-09-15:
Go to LowPower.cpp, find this lines
`ISR (WDT_vect)
{
	// WDIE & WDIF is cleared in hardware upon entering this ISR
	wdt_disable();
}`
and comment it

ICPete, 2024-09-02:
I know this thread is 4 years stale, but thank you, eldiamond2; your solution helped me!!

jrleeman, 2024-10-31:
Ditto - very helpful - in my case a watchdog library was causing the conflict. Thanks @eldiamond2