rocketscream/Low-Power #84: No sleep_forever on Atmega328p

mrguen, 2019-04-13:
I just tried to use the code on Atmega328p. It used to work on Atmega168p, but now it was not working anymore. I am not sure this is an MCU related issue, but clearly the WDT is still enabled when entering sleep and so the MCU wakes up after a period (one second in my case). 

To test this hypothesis I added wdt_disable() before calling LowPower.powerDown(SLEEP_FOREVER, ADC_OFF, BOD_ON); and indeed it does sleep forever now which was the objective.

In LowPower.cpp wdt_disable()  is called only in the ISR (WDT_vect). So apparently it would need to be initialized in any case otherwise it behaves depending on the last configuration.

rocketscream, 2019-04-14:
Hi mrguen,
This is odd because prior to your report, I just tested 10 boards of ATMega328P, they worked accordingly. May I know what library revision and avr-gcc revision you are using?

mrguen, 2019-04-15:
I use my fork https://github.com/mrguen/Low-Power based on your version 1.8
Arduino IDE 1.8.9

If I was not explicit enough: SLEEP _FOREVER works when called first. It does not work when called after calling SLEEP_xxS because SLEEP_xxS activates wdt and apparently it is never deactivated.

rocketscream, 2019-04-15:
I just tried my own branch and it works just fine here:

```
// **** INCLUDES *****
#include "LowPower.h"

void setup()
{
   pinMode(13, OUTPUT);
   digitalWrite(13, HIGH);
   LowPower.powerDown(SLEEP_1S, ADC_OFF, BOD_OFF);  
   digitalWrite(13, LOW);
}

void loop() 
{
    // Enter power down state for 8 s with ADC and BOD module disabled
    LowPower.powerDown(SLEEP_FOREVER, ADC_OFF, BOD_OFF);  
    
    // MCU should not wake up here
}
```
With the above code, you should see the LED turn on for 1 s and then off.

If you look at the code in the WDT ISR, the watchdog is disabled:

```
ISR (WDT_vect)
{
	// WDIE & WDIF is cleared in hardware upon entering this ISR
	wdt_disable();
}
```

mrguen, 2019-04-16:
Maybe I don't get it but of course the led remains low since it is positionned to low... Maybe you should add a digitalWrite(13, HIGH ); after callling SLEEP_FOREVER....

manosv, 2019-04-16:
of course it stays low because in the loop function he has to put digitalWrite(13, HIGH ) after LowPower.powerDown(SLEEP_FOREVER, ADC_OFF, BOD_OFF);

rocketscream, 2019-04-16:
Sorry guys, forgot to mention that I also used a meter for the setup (to double check current measurement) so even with the digitalWrite(13, HIGH); after LowPower.powerDown(SLEEP_FOREVER, ADC_OFF, BOD_OFF); it doesn't wake up. Measurement is also spot on. What AVR board package version you are using on the Arduino IDE? Mine is the latest of 1.6.23.

mrguen, 2019-04-16:
Where do you get the AVR package version? / I just installed the latest Arduino version.

II have seen a couple of posts (as well as https://www.nongnu.org/avr-libc/user-manual/group__avr__watchdog.html) complaining about wdt staying enabled and also that the code should be 

MCUSR = 0;
wdt_disable( );

But I really don't get the details.

mrguen, 2019-04-17:
I tried with MCUSR = 0; but wdt is still activated.

At the moment the code that works is 

if (deviceOFF) {
	INA226_USB.configure(INA226_AVERAGES_16, INA226_BUS_CONV_TIME_1100US, 
 	INA226_SHUNT_CONV_TIME_1100US, INA226_MODE_POWER_DOWN);
	INA226_VIN.configure(INA226_AVERAGES_16, INA226_BUS_CONV_TIME_1100US, 
        INA226_SHUNT_CONV_TIME_1100US, INA226_MODE_POWER_DOWN);
	oled.ssd1306WriteCmd(SSD1306_DISPLAYOFF);
	
	wdt_disable(); 
	LowPower.powerDown(SLEEP_FOREVER, ADC_OFF, BOD_ON);
	
        oled.ssd1306WriteCmd(SSD1306_DISPLAYON);
	deviceOFF = false;
	}
else {
	LowPower.powerDown((period_t)period, ADC_OFF, BOD_ON);
	}

Configuration:
* AVR Board 1.6.23
* Device signature = 0x1e950f (probably m328p)
* Fuse bytes set by downloading the bootloader for Nano board
* But no bootloader, the code is downloaded by ISP

rocketscream, 2019-04-18:
I would advise to use a more standard setup: bootloader and no other libraries. That way we can eliminate other possible factors. 

I have check the assembly listing of the generated code I posted earlier and it is very clear that the wdt_disable() is called within the ISR_WDT.

mrguen, 2019-04-21:
I can't use a bootloader since this board does not have USB implemented and the libraries are needed. For the moment I will keep wdt_disable(); before calling powerDown.

closed 2020-09-08 by rocketscream