rocketscream/Low-Power #85: Using Low-Power with a hardware watchdog?

Bambofy, 2019-06-03:
Hi, i need to ping a watchdog frequently to keep the Arduino running.

Am i ok to loop say, 10 times, and perform a SLEEP_60MS multiple times, pinging the watchdog after each sleep?

Bambofy, 2019-06-03:
Like this:
`  int numberOfLoopsToPerform = (int)numberOfLoopsInDelay;
  
  for (int i = 0; i < numberOfLoopsToPerform; i++)
  {
    Pin_Interface_WatchdogHeartbeat();
    
    LowPower.powerDown(SLEEP_1S, ADC_OFF, BOD_OFF);
  }`

robervaltaylor, 2020-08-30:
Related to #89

mariovaldez, 2021-01-02:
Yes, that will work, I use similar code with external watchdog timers (like the TPS3828 or a second MCU). I just would just make sure the external watchdog signal is send not only before sleeping but also on wake up, so your code would be: 

```
  int numberOfLoopsToPerform = (int)numberOfLoopsInDelay;
  for (int i = 0; i < numberOfLoopsToPerform; i++)
  {
    Pin_Interface_WatchdogHeartbeat();
    LowPower.powerDown(SLEEP_1S, ADC_OFF, BOD_OFF);
  }
  Pin_Interface_WatchdogHeartbeat();
```