rocketscream/Low-Power #98: Define the sleep time.

alessandrolecce, 2020-04-04:
Is it possible to define sleep time? 
I need to sleep the Arduino for 5 minutes. How can I implement it in LowPower.idle() ?

Adrian363, 2020-05-06:
Hi, 
I want to make the same thing as you. Have you found a solution to increase sleeping time?

alessandrolecce, 2020-05-06:
Not, yet, I will work on it by the end of this month

Inviato da Outlook Mobile<https://aka.ms/blhgte>

ziplock9000, 2020-05-10:
This is exactly what I was coming to request too :)

zvoniimiir, 2020-06-05:
Have you tried powerDown? sleepCounter represents 8s periods (due to SLEEP_8S). So 5 minutes would be sleepCounter = 38 

`for (int sleepCounter = 38; sleepCounter > 0; sleepCounter--) 
  {
    LowPower.powerDown(SLEEP_8S, ADC_OFF, BOD_OFF);
  }`

ziplock9000, 2020-06-05:
> Have you tried powerDown? sleepCounter represents 8s periods (due to SLEEP_8S). So 5 minutes would be sleepCounter = 38
> 
> `for (int sleepCounter = 38; sleepCounter > 0; sleepCounter--) { LowPower.powerDown(SLEEP_8S, ADC_OFF, BOD_OFF); }`

That still means it's wakes up every multiple of 8 seconds. While this would be sufficient for most use cases, but it does not have a single defined sleep time with no wakes like the title mentions.

rocketscream, 2020-09-08:
The timer is based on the watchdog timer which has discrete period that cannot be changed. Using the `for` loop is the only way here without any external RTC IC or upgrade to the newer [megaAVR 0-series MCU that has real time counter](https://github.com/rocketscream/RocketScream_RTCAVRZero).

closed 2020-09-08 by rocketscream