rocketscream/Low-Power #100: idle() function doesnt seem to restore UART/Serial when using USART0_OFF

tjko, 2020-05-23:
If using UART/Serial port it works fine until first call to idle() after that serial is not working (it's sending just garbage). It would seem like UART settings get lost duing low power state and it doesn't get restored after resuming?

Also, similar to issue https://github.com/rocketscream/Low-Power/issues/36
if calling with "USART0_ON", serial port does not get messed up, but similar to using "TIMER0_ON" call to idle() will return within a millisecond (SLEEP_x) option doesn't work...

AnnyCaroline, 2021-04-25:
I'm also having this problem.  But, for me, even using `USART0_ON`, the behavior is not as expected. The messages are displayed on the serial, but very quickly. In the example below, the LED connected to pin 13 also switches very quickly, not respecting the 4s.

```
#include "LowPower.h"

void setup() {
  Serial.begin(9600);
  Serial.println("BEGIN");

  pinMode(13, OUTPUT);
  digitalWrite(13, LOW);
}

void loop() {
  LowPower.idle(SLEEP_4S, ADC_OFF, TIMER2_OFF, TIMER1_OFF, TIMER0_OFF, SPI_OFF, USART0_ON, TWI_OFF);

  Serial.println("4s - ON");
  digitalWrite(13,HIGH);

  LowPower.idle(SLEEP_4S, ADC_OFF, TIMER2_OFF, TIMER1_OFF, TIMER0_OFF, SPI_OFF, USART0_ON, TWI_OFF);

  Serial.println("4s - OFF");
  digitalWrite(13,LOW);
}
```

Before starting to use `Serial.println`, the LED was blinking at the correct interval.

eddyp, 2021-07-29:
@AnnyCaroline I think the issue which you're seeing is because the Serial is actually itself generating interrupts, so the sleep cycles are stopped prematurely.

I saw this @rocketscream library doesn't have the latest versioned release in Ardiuino IDE, so I switched to @LowPowerLab's fork which seems to merge all changes from this repo and some fixes, plus is maintained.

Ironically the latest code here has a fix for the correct restauration of timer2, part of 1.81/1.8.1, but the latest release in Arduino IDE is 1.60/1.6.0