rocketscream/Low-Power #107: Issue - ATmega328 @ 3.3V 8MHz Int RC Osc

pranjal-joshi, 2020-11-26:
Hello,
While using the `LowPower` library for Arduino Pro Mini with ATmega328 @ 3.3V 8MHz Int RC Osc, the CPU doesn't wake up with a watchdog timer.
My code snippet is as follow:
```
    attachInterrupt(digitalPinToInterrupt(PIN_CAN_INTR), isr_can, LOW);
    attachInterrupt(digitalPinToInterrupt(PIN_RS232_INTR), isr_rs232, LOW);
    Serial.flush();
    Serial.end();
    LowPower.powerDown(SLEEP_FOREVER, ADC_OFF, BOD_OFF);
```
**All of the following parameters works like `SLEEP_FOREVER` and no watchdog reset!**
```
enum period_t
{
	SLEEP_15MS,
	SLEEP_30MS,
	SLEEP_60MS,
	SLEEP_120MS,
	SLEEP_250MS,
	SLEEP_500MS,
	SLEEP_1S,
	SLEEP_2S,
	SLEEP_4S,
	SLEEP_8S,
	SLEEP_FOREVER
};
```

rocketscream, 2020-11-26:
Please post entire code. The external interrupt does have timing limitation (look at the datasheet) in the sense it needs to of certain minimum clock width for the ISR to determine what fired up. I believe the internal oscillator is even slower to start up compared to crystal. So, after oscillator start up has completed, your LOW signal might already went up HIGH. Depending on your code, it probably might look like sleep-wake-sleep continuously with the wake up period a short period.

pranjal-joshi, 2020-11-26:
I will post the new results after changing my design to 5V 16MHz

BECS3D, 2021-06-10:
Please update this as I am also having this issue and would like it to be resolved if there is a known solution. Thanks

pranjal-joshi, 2021-06-20:
@BECS3D I decided to implement wake-up using the raw method suggested in the datasheet by configuring the registers.