rocketscream/Low-Power #108: The function of wakeUpPin?

sxontheway, 2020-12-03:
Hi, I am wondering that in "powerDownWakeExternalInterrupt.ino", what is the function of "wakeUpPin = 2" ? Since the variable wakeUpPin does not be revoked anywhere else during the loop(). Does it means that any voltage change on the pin can wake up the system?

percz, 2020-12-07:
Interrupts interrupt whatever loop() is doing. 

You can have a long list of lines in a function, but if your interrupt is triggered the processor will jump to whatever function is linked to it ( in this example wakeUp() )

Arduino actually does this for millis(). No matter what else your own loop code is doing, it will very quickly stop at exactly a certain time, update the millis() value a little then return to where it left your own code.

sxontheway, 2020-12-08:
> Interrupts interrupt whatever loop() is doing.
>
> You can have a long list of lines in a function, but if your interrupt is triggered the processor will jump to whatever function is linked to it ( in [...]

Thanks for your answer. I searched online and found that the code in line22 of the `powerDownWakeExternalInterrupt.ino` is usually written as `attachInterrupt(digitalPinToInterrupt(interruptPin), wakeUp, LOW);`. 

I find the definition:
`#define digitalPinToInterrupt(p) ((p) == 0 ? 2 : ((p) == 1 ? 3 : ((p) == 2 ? 1 : ((p) == 3 ? 0 : ((p) == 7 ? 4 : NOT_AN_INTERRUPT)))))`

Does it means that for Arduino (AVR versions), certain pins have certain interrupt numbers? So if we use `attachInterrupt(0, wakeUp, LOW);` actually we define pin3 as the interrupt pin?