rocketscream/Low-Power #109: the clock ( millis() ) seems not to work in a wakeup function

dan-bel, 2021-01-26:
Hi,

I experienced that the millis() is always returning the same value when run in a function which is triggered after waking up.

Unfortunately, I lack the skills to dig deeper :(

Attached a sample code.
After 10 sec, the Arduino will go to sleep and is woken up by connecting pin 2 to ground.
The function wakeUp() is run, and it seems that time stays still.
Tested on Arduino nano, and LowPower version 1.81

Thank you

Dan

rocketscream, 2021-01-31:
All clock are stopped during power down. `millis()` only run during active state.

dan-bel, 2021-01-31:
that makes sense, but I would expect the wakeup function to be an active state since things are happening?

rocketscream, 2021-02-01:
The `millis()` stops during sleep and upon waking up it will continue to run, hence the `millis()` will stop to count during the sleep period. If just prior to sleep the `millis()` count is 1000 ms, it will be 1000 ms upon waking up despite sleeping for example 8 seconds.

dan-bel, 2021-02-05:
Thank you, I now realized that time never runs in interrupt handlers, so it would also not run here.
Thank you for your input and clarification.

closed 2021-02-05 by dan-bel