rocketscream/Low-Power #123: Missing license?

GFMsoft, 2022-11-08:
Hey rocketscream,

firstly i wanna thank you for your great work.
I just want to know under what kind of a license your software is distributed?

Can't find any information in your code nor on Github.

Thank you.

slatian, 2024-03-31:
I second that, please add a license. 'm pretty sure you intend for the code to be reusable but unless you add a license [nobody really is allowed to](https://reuse.software/faq/#no-copyright-license).

GFMsoft, 2024-04-02:
I found a license in the header documentation.
`This library is licensed under Creative Commons Attribution-ShareAlike 3.0`

Would be nice inside the GitProject :)

slatian, 2024-04-02:
Thank you! Yes, License file with the license and/or a mention in the README